microbotsio/CodeCell #1: Sensor init issue

quinm0, 2024-10-23:
Hi, I plan on looking into this more myself but I just want to document the first issue I got

I was using the example from the library

```
#include <CodeCell.h>

CodeCell myCodeCell;

/*
 * Overview:
 * This code shows you how easy it is to set up and run the CodeCell. You just needs to specify which sensors to enable by 
 * passing the appropriate macros to the `Init()` function and use the `Run()` function to handle the power and battery management.
 *
 **** Initialization (`Init` function):
 * 
 * To initialize the CodeCell, use the `myCodeCell.Init()` function with one or more of the predefined macros. Each macro
 * corresponds to a specific sensing function. Here are the available macros:
 * 
 * - `LIGHT`                          // Enables Light Sensing
 * - `MOTION_ACCELEROMETER`           // Enables Accelerometer Sensing
 * - `MOTION_GYRO`                    // Enables Gyroscope Sensing
 * - `MOTION_MAGNETOMETER`            // Enables Magnetometer Sensing
 * - `MOTION_LINEAR_ACC`              // Enables Linear Acceleration Sensing
 * - `MOTION_GRAVITY`                 // Enables Gravity Sensing
 * - `MOTION_ROTATION`                // Enables Rotation Sensing
 * - `MOTION_ROTATION_NO_MAG`         // Enables Rotation Sensing without Magnetometer
 * - `MOTION_STEP_COUNTER`            // Enables Walking Step Counter
 * - `MOTION_STATE`                   // Enables Motion State Detection
 * - `MOTION_TAP_DETECTOR`            // Enables Tap Detector
 * - `MOTION_ACTIVITY`                // Enables Motion Activity Recognition
 * 
 * Example Usage:
 * - `myCodeCell.Init(LIGHT);`                                      // Initializes Light Sensing
 * - `myCodeCell.Init(LIGHT + MOTION_ROTATION);`                    // Initializes Light Sensing and Angular Rotation Sensing
 * - `myCodeCell.Init(LIGHT + MOTION_ROTATION + MOTION_STATE);`     // Initializes Light Sensing, Angular Rotation Sensing, and State Detection
 * 
 * Note: You can combine multiple macros using the `+` operator to initialize multiple sensors.
 * 
 **** Running the Code (`Run` function):
 * 
 * Call the `myCodeCell.Run()` function in the `loop()` to manage battery and power. This function returns true every 100ms and
 * also handles the onboard LED to indicate power status. When the battery voltage falls below 3.3V, the LED will blink red 10 times 
 * and then go into Sleep Mode until the USB cable is connected for charging. While charging, the CodeCell will shut down the application, 
 * light the LED blue, and wait until the battery is fully charged. Once fully charged, it will start a breathing-light animation with a speed
 * corresponding to the proximity distance. The LED will shine green when powered by the battery and blue when powered via USB.
 * 
 **** Reading Sensor Data:
 * 
 * After initializing the sensors, you can use the following functions to read data from them:
 * 
 * Sensor Read Functions:
 * 
 * - `Light_ProximityRead()`                                            // Reads the proximity value from the light sensor
 * - `Light_WhiteRead()`                                                // Reads the white light intensity from the light sensor
 * - `Light_AmbientRead()`                                              // Reads the ambient light intensity from the light sensor
 * - `Motion_TapRead()`                                                  // Reads the number of taps detected (tap = 1, no tap = 0)
 * - `Motion_StateRead()`                                               // Reads the current state (On Table = 1, Stationary = 2, Stable = 3, Motion = 4)
 * - `Motion_ActivityRead()`                                            // Reads the current activity (Driving = 1, Cycling = 2, Walking = 3/6, Still = 4, Tilting = 5, Running = 7, Climbing Stairs = 8)
 * - `Motion_AccelerometerRead(float &x, float &y, float &z)`           // Reads acceleration data along the x, y, and z axes
 * - `Motion_GyroRead(float &x, float &y, float &z)`                    // Reads rotational velocity data along the x, y, and z axes
 * - `Motion_MagnetometerRead(float &x, float &y, float &z)`            // Reads magnetic field strength data along the x, y, and z axes
 * - `Motion_GravityRead(float &x, float &y, float &z)`                 // Reads gravity vector data along the x, y, and z axes
 * - `Motion_LinearAccRead(float &x, float &y, float &z)`               // Reads linear acceleration data along the x, y, and z axes
 * - `Motion_StepCounterRead(uint16_t &x)`                              // Reads the number of steps counted
 * - `Motion_RotationRead(float &roll, float &pitch, float &yaw)`       // Reads angular rotational data (roll, pitch, yaw)
 * - `Motion_RotationNoMagRead(float &roll, float &pitch, float &yaw)`  // Reads angular rotational data without magnetometer
 *
 * Example Usage:
 * - `uint16_t proximity = myCodeCell.Light_ProximityRead();`
[... 85 lines cut ...]
3fc993b0: 0x202e2e73 0x776f6e69 0x2e2e7265 0x72500a0a 0x6d69786f 0xad427469 0x20c93ae3 0x35e6014e
3fc993d0: 0xfde87a79 0x768916f2 0x7ded35f9 0x6364ab25 0x0188fd72 0x1c710fc8 0xc320b46d 0x1b200100
3fc993f0: 0xf0eafe6b 0xefafb3b7 0xbf2bbeff 0xd3fd75f9 0x687c063f 0x74fc789e 0xc4084012 0xed19a836
3fc99410: 0x6d33cb13 0x5abbf116 0xdfca3dfe 0x81094d4e 0x3fcb1b03 0x965e8a10 0xd59d8973 0x822f7e6d
3fc99430: 0xbdffa339 0x6fdff7f1 0x7bf7f32f 0x667fafad 0x24330707 0x48470868 0xa2b53d35 0x44a9c44c
3fc99450: 0x9f7ffff5 0xf9dbaf16 0x37d7d311 0xff4ff312 0x2d2e7563 0x750209c5 0x31803043 0x5a433f00
3fc99470: 0xf71d97b0 0xfc3a6e2e 0xbffde733 0xa3b10b7e 0x25a08220 0x2bc80a72 0x4b08094c 0xa24010ba
3fc99490: 0xf6b7ec90 0x9f71bfc0 0x512e72fe 0xd1771fa7 0x832628d0 0xbaad5678 0x0000007c 0xabba1234
3fc994b0: 0x00000070 0x00000100 0x00000100 0x00000002 0x40385b0c 0x4038618c 0x40385a42 0x403866d4

ELF file SHA256: 1135b25dfb9d017e

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1058
load:0x403cc710,len:0x8a8
load:0x403ce710,len:0x2fac
entry 0x403cc710

```

microbotsio, 2024-10-24:
Hi Quin

Thanks for reaching out and for your recent order. From your order number I
can see that you ordered the Light sensor not the Light + Motion sensor

This means that the CodeCell is not finding the motion sensor when you
added "+ MOTION_ACCELEROMETER", as it is not present on the board -so it
keeps resetting indicating the error

Let me know if you have further questions.

[image: Logo]
Microbots Team
Miniaturized Robotics + Simplified Electronics
email: ***@***.***
www.microbots.io
[image: Twitter icon] <https://twitter.com/microbots_io>  [image: Instagram
icon] <https://www.instagram.com/microbots.io/>

microbotsio, 2024-10-24:
Hi Quin

Are you using the CodeCell Light or CodeCell light+motion?

This means that the CodeCell is not finding the motion sensor when you
added "+ MOTION_ACCELEROMETER", as it is not present on the board -so it
keeps resetting indicating the error

Let me know if you have further questions.

closed 2025-02-08 by microbotsio