microbotsio/CodeCell #3: When connected to USB with battery connected, application does not run

divbell, 2024-11-01:
I've noticed that when I have the battery connected and the USB connected, the application shuts down and it goes into battery charging mode and never seems to return to the application, even when the battery is apparently charged (or charged enough).

Is there a way to keep the application running while connected to USB with a battery connected, and use excess power to top-off the battery?

microbotsio, 2024-11-05:
This can be done, however the Run function currently put the device in sleep mode while the battery is connected and charging via usb, turning off all gpio pins. We will consider adding a way to disable this in the next release

greiginsydney, 2025-01-31:
> We will consider adding a way to disable this in the next release

That would be helpful, as it would allow us to debug via the USB/serial port

microbotsio, 2025-02-25:
Just released a new software release v1.2.7 that introduced the USBChargeState() function to optional keep CodeCell awake while charging. Here's how you can use it in the setup:

`void setup() {
  Serial.begin(115200); // Set Serial baud rate to 115200 - Ensure Tools/USB_CDC_On_Boot is enabled if using Serial

  myCodeCell.Init(LIGHT); // Initializes Light Sensor
  
  myCodeCell.USBChargeState(1);
}`

Please note when enabling this option the battery will automatically keep charging and the LED will blink blue to indicate the USB power

closed 2025-02-25 by microbotsio