microbotsio/CodeCell #8: Interrupt connected?

braddo99, 2025-02-15:
Hey, the schematic shows the BNO085 interrupt pin is not connected, is that true?

I'm trying to use the CodeCell in an application that seems to require/use the interrupt for the BNO085, it would be great if that is a mistake and it actually is connected :-) 

Thanks, cool product!

microbotsio, 2025-02-15:
Unfortunately the interrupt pin was not feasible to include it, as we ran
out of pins. However you can poll the sensor and even go to sleep, like
shown in this example:
https://github.com/microbotsio/CodeCell/blob/main/examples/Light_SleepExample/Light_SleepExample.ino

Let us know if you have any further questions

braddo99, 2025-02-15:
Thanks for the reply. Sorry for a couple more questions on this topic because I don't control the software for the project and if i can't hook up the interrupt I can't use your boards, and I now own 4 :-(
Is there any place on the board where the int pin for the BNO physically exists (other than under the chip pad) so that I might jumper it to an open gpio?
The Arduino IDE does work and I have run some of your examples, great! But, I haven't been able to upload the code to the board at all yet using VS Code and Platformio. Do you have any instructions for that?
Thanks again,
Brad

microbotsio, 2025-02-17:
Hi Brad We understand the frustration and appreciate your feedback :)

Unfortunately the interrupt pin could not be connected due to the small size of the module. So at the moment we suggest polling the sensor, but we are considering other solution for future versions of the board to make it more accessible. 

You can use Platform IO for any esp32 device, including the CodeCell. But at the moment the library is not compatible because PlatrformIO does use the latest/same libraries for the esp32 board manger, as the Arduino IDE , so its fails to build and initialize the timers. So for now we suggest using the Arduino IDE to use the codecell library, but we will take note of your request and try to push it for later this year

braddo99, 2025-02-18:
I have to use VS Code and PltformIO because that is the platform the code I am using is built on. However, I could probably adjust the version of the esp32 libraries that are used etc. But I do think I require a board/configuration file for your device to take into account how the pins (even ones I"m not using) are connected. Do you have any information on that? It may be possible to tack a wire onto the pad for the interrupt, luckily how you have it oriented on the board makes it relatively accessible. Not easy since it's just a pad underneath, but I can see some exposed solder paste, maybe I can get a micro wire under there with the microscope. It would be great if you could connect the interrupt in a future version, I'm not sure about the size constraint - if you put a 0 ohm resistor on so that a user could switch out the light sensor or even just a pad so that one could solder to it, that would be useful. Maybe even set some fuses in the firmware that would allow a user to point IO to the desired devices on the board.