microbotsio/CodeCell #9: CodeCell restart Issues with Run() Function

JamesMegariotis, 2025-02-23:
Hi,

I just purchased a few codecells to test for an IOT sensor project we have been working on. I am really struggling to understand the logic behind some of the power state logic in the [Run()](https://github.com/microbotsio/CodeCell/blob/219996ab81475c8aa5fdf79adadb00b16f6efbb9/src/CodeCell.cpp#L379) function in CodeCell.cpp. I am experiencing random restarts when on battery power. I have had to implement my own helper functions and completely avoid using the codecell library. I am now wondering if there is a reason this logic is implemented this way.

It appears, as best I can understand the logic, is that the ESP32 is put to sleep via USBSleep() function when the battery is charging. Is there a design limitation that necessitates this behavior? Or is this just a matter of preference? I only ask since this seems like a power save function, but during charging, power consumption of the main program should not be much of a concern, so I am wondering if there is another motive behind this decision.

It also seems to issue esp_restart() ([line 445](https://github.com/microbotsio/CodeCell/blob/219996ab81475c8aa5fdf79adadb00b16f6efbb9/src/CodeCell.cpp#L445) and[ line 420](https://github.com/microbotsio/CodeCell/blob/219996ab81475c8aa5fdf79adadb00b16f6efbb9/src/CodeCell.cpp#L420)) when the TI charge controller IC is in the charge state. I am sure that these restarts appear to be called at times that are not appropriate based on my testing. The longest my code runs without a restart is 5 mins on battery power with nominal battery voltage and no USB plugged in. This appears to be a bug of some sort causing esp_restart() to be called even with the usb unplugged and a valid battery charge is present. Could this possibly be due to the [USB_VOLTAGE](https://github.com/microbotsio/CodeCell/blob/219996ab81475c8aa5fdf79adadb00b16f6efbb9/src/CodeCell.h#L16) threshold being set to 4.1V. I beleive the charger IC is charging to above that voltage? I am suspicious the issue lies in the following [lines of code](https://github.com/microbotsio/CodeCell/blob/219996ab81475c8aa5fdf79adadb00b16f6efbb9/src/CodeCell.cpp#L436C8-L446C12) running erroneously and may be triggered by a high battery voltage. 

I have a similar question for this feature. Is there a design reason behind this decision to restart the esp32 when a USB is plugged or unplugged?

Also any clarity you could provide on the [charge state constants](https://github.com/microbotsio/CodeCell/blob/219996ab81475c8aa5fdf79adadb00b16f6efbb9/src/CodeCell.h#L12C1-L15C31) would be appreciated. I think I understand what some of these states represent, but not 100% sure on others, specifically POWER_INIT and POWER_BAT_CHRG_FULL. Do these represent USB power being applied and the completion of charging?

I would love to use this library as it simplifies the interface significantly, but it seems to be at odds with the purpose of our sensor firmware. Is it required to execute the run() function in the arduino loop? I initially thought it was, but now I am beginning to question that assumption.

Any insight you have on these questions as well as the unexpected reboot issue on battery power would be appreciated.

Thanks,
James

microbotsio, 2025-02-24:
Hi James

Thanks for reaching out :) Can you share your code in the
initialization and loop? And also verify that you are using the latest
library version?

Calling the Run() function in the main loop should not trigger any resets -
in software version v1.2.4 (and below) a soft reset was being called if a
sensor is read but not initialized, but in v1.2.5 this was updated,
outputting an error on the serial monitor instead. So it seems very
strange that the CodeCell is resetting itself.

Regarding the Power Handling, on v1.2.6 a soft reset is being applied in
the Run() function in the cases listed below:

   - CodeCell is Battery Powered & a USB cable is plugged in for charging
   - CodeCell is Battery Powered & a USB cable is plugged in for charging -
   during the battery low indication warning
   - If Serial connection is lost but USB power still active

The  USB_VOLTAGE is the threshold we use to indicate if the cable is
powered via USB or battery. The output voltage when USB powered, can vary
around 0.2V from one chip to another, so from our testing 4.1V is a good
limit.

POWER_INIT is being used internally as an initialization flag, for us to
detect which power stage the CodeCell is in, whether its battery power, usb
powered, or in charging stage.
POWER_BAT_CHRG_FULL indicates that the battery is fully charged

Looking forward for your response

Microbots Team

JamesMegariotis, 2025-02-24:
Hi,

I am currently using codecell library v1.2.6 and was using this version throughout the issues I encountered. We have a larger project that is not currently publicly available for use in AR/VR simulations, but this code supports a variety of esp32 based microcontrollers, IMUs and battery management systems. We added a very minimal amount of code to support the codecell microcontroller. This overall project has run in a stable manner on the other supported microcontrollers. The codecell also has nothing but the battery on it. So no peripheral hardware or other drivers are running. Only what is included on a stock codecell with IMU. The only unique code that was added for the code cell is below. 

My initialization for the code cell in setup() was the following: 
`myCodeCell.Init(MOTION_ROTATION_NO_MAG);`

In the loop() I was doing just doing the following:
```
// codecell IMU read code
if (myCodeCell.Run(50)) {  //Run every 50Hz
    myCodeCell.Motion_RotationNoMagRead(roll_f, pitch_f, yaw_f);
}
```

Roll pitch and yaw are then packaged and sent in our data packets. The only other call to the code cell library I used was the BatteryRead() function to populate battery voltage in our data packets and for display on a web server. 

Upon removing the code cell library and using the adafruit BNO085 library for IMU and manually reading the battery voltage and charger IC state through analog and digital read, the code runs stable for hours, but any attempt to use the above few lines of code seems to lead to stability issues and resets within minutes when on battery power only. I assume it must either be sleeping and waking up or calling esp_restart, since all wifi connections are lost and our setup() code seemingly runs again, resetting certain timers/counters that only reset on startup.

Is calling the Run() function even required to be able to read the IMU throught this library? Because it seems my issue is definitely coming from there. I am also not super fond of the esp32 sleeping while charging. Is there a reason that run implements this? I only wonder because saving power while plugged in via usb seems like it would not be a major concern unless there was a hardware limitation or reason for doing so.

-James

microbotsio, 2025-02-25:
Hi James

Thanks for the clarification :)

We will try to replicate the bug and get back to you

Microbots Team

microbotsio, 2025-02-26:
Hi James

So far, we have not been able to replicate the issue, but we suspect it is related to the tolerance of the charging chip and ADC. We have identified possible root causes that could explain the behavior you observed - so we have released an update to the v1.2.7 library to prevent unnecessary resets related to charging. 

Could you update to this new version and confirm whether the issue still persists?

microbotsio, 2025-03-15:
Hi James

Just checking if you have any update on this

microbotsio, 2025-04-09:
Since we haven’t heard back, we’re assuming the issue has been resolved.
If you have any further questions, please don’t hesitate to reopen the issue

closed 2025-04-09 by microbotsio