microbotsio/CodeCell #11: Function usage

Nakupenda-star, 2025-03-30:
The code `timerBegin(100000)` used in CodeCell.cpp reported an error during the compilation process. The function prototype is `hw_timer_t* timerBegin(uint8_t timer_num, uint16_t divider, bool count_up)`. Is this due to the version you used in the past, or did I misunderstand?

microbotsio, 2025-03-31:
Hi

Thanks for reaching out!

Can you confirm you have the latest esp32 board-library installed? The
CodeCell support v3.1.1 and above

Thanks
Team Microbots

microbotsio, 2025-04-04:
> I used the VCCode platformIO development environment to recompile your
> project and found errors. What should I pay attention to in the current
> development environment? How can I correctly create an esp32-C3 project?

First, could you confirm if your device works with the Arduino IDE?

Our official CodeCell library is written in C++ and is primarily designed
for the Arduino IDE environment. If you’re using PlatformIO, it’s possible
you might need to write some custom code to interface properly with the
onboard sensors and peripherals, especially if PlatformIO doesn’t have the
latest ESP32 board libraries installed.

For reference, on Arduino IDE we recommend using ESP32 board support
package v3.1.1 or above, which includes the latest definitions and support
for the ESP32-C3.

As for the timerBegin() issue - The error likely comes from an outdated
board package, since the syntax and function signatures were updated in
recent ESP32 core versions. In the Arduino IDE with the latest version, the
code compiles as expected. So if you cannot update the esp32 package
library you could either not use the timer or modify it, to work on the
specific version you are using.

microbotsio, 2025-04-22:
Hi

Is there an update on this issue?

microbotsio, 2025-04-30:
As there’s been no response for several weeks, I’m closing this issue for now.

If the problem persists or you have more information to add, feel free to reopen it or open a new issue.

Thank you.

closed 2025-04-30 by microbotsio