microbotsio/CodeCell #12: loop reboot

openmotion, 2025-04-04:
hello randomly i have this problem with the board
do you think it's an hardware problem ?

>> CodeCell v1.2.7 Booting.. 
>> Initializing Sensors: E (2496) i2c.master: I2C transaction timeout detected
E (2497) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up

>> Error: Motion Sensor not found - Check Hardware

microbotsio, 2025-04-04:
Hi

Thanks for reaching out! There was an issue on some devices, related to
this bug on software v1.2.7

Can you check if the issue occurs with the latest software update? v1.2.8

Thanks
Team Microbots

openmotion, 2025-04-04:
> Hi
> 
> Thanks for reaching out! There was an issue on some devices, related to
> this bug on software v1.2.7
> 
> Can you check if the issue occurs with the latest software update? v1.2.8
> 
> Thanks
> Team Microbots
> […](#)

thanks i updated but same problem

```
>> CodeCell v1.2.8 Booting.. 
>> Initializing Sensors: E (3167) rmt: rmt_new_tx_channel(269): not able to power down in light sleep

>> Error: Motion Sensor not found - Check Hardware
```

microbotsio, 2025-04-04:
Can you share the code? It seems the device is going to sleep

openmotion, 2025-04-04:
i used the basic example, commented you the wifi part

```
#include <WiFi.h>
#include <CodeCell.h>
#include <BLEServer.h>
#include <BLE2902.h>
CodeCell myCodeCell;

const char* ssid = "xxx"; // Replace with your Wi-Fi name
const char* password = "xxx"; // Replace with your Wi-Fi password

float x = 0.0;  // Magnetic field on X-axis
float y = 0.0;  // Magnetic field on Y-axis
float z = 0.0;  // Magnetic field on Z-axis

void setup() {
  // put your setup code here, to run once:
    Serial.begin(9600);
  while(!Serial);

// WiFi.begin(ssid, password); // Connect to Wi-Fi

  //  Serial.print("Connecting to Wi-Fi");
   // while (WiFi.status() != WL_CONNECTED) {
    //    delay(500);
     //   Serial.print("."); // Print dots while connecting
    //}
    
  //  Serial.println("\nConnected!");
//    Serial.print("IP Address: ");
//  Serial.println(WiFi.localIP()); 
   Serial.println("start magnetometer");
    myCodeCell.Init(MOTION_MAGNETOMETER); // Initialize the magnetometer
}

void loop() {
  // put your main code here, to run repeatedly:
   //Serial.println("start");

  if (myCodeCell.Run(10)) { // Read at 10Hz (every 100ms)
    myCodeCell.Motion_MagnetometerRead(x, y, z); // Get magnetometer readings

     Serial.print("Mag X: "); Serial.print(x);
     Serial.print("  Y: "); Serial.print(y);
     Serial.print("  Z: "); Serial.println(z);
  }

}

```

microbotsio, 2025-04-07:
Hi

We are trying to replace the issue with your board - Can you confirm how
frequently it is happening with the v1.2.8 software? And if you have any
external sensors/modules connected to the I2C bus?

Thanks
Team Microbots

openmotion, 2025-04-07:
> Hi
> 
> We are trying to replace the issue with your board - Can you confirm how
> frequently it is happening with the v1.2.8 software? And if you have any
> external sensors/modules connected to the I2C bus?
> 
> Thanks
> Team Microbots
> […](#)

it's happening evertytime now
`>> CodeCell v1.2.8 Booting.. 
>> Initializing Sensors: E (2915) i2c.master: I2C transaction timeout detected
E (2915) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up
E (2928) i2c.master: I2C hardware timeout detected
E (2928) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up

>> Error: Motion Sensor not found - Check Hardware`

microbotsio, 2025-04-07:
Understood but can you confirm if you have any external sensors/modules
connected to the I2C bus? or if the sensor was damaged in some way?

openmotion, 2025-04-07:
> Understood but can you confirm if you have any external sensors/modules
> connected to the I2C bus? or if the sensor was damaged in some way?
> […](#)

nothing is connected to the I2C bus, i think the sensor is damaged
can you send me another board ?

microbotsio, 2025-04-07:
Can you send us your order details by email on ***@***.***

openmotion, 2025-04-07:
> Can you send us your order details by email on ***@***.***
> […](#)

yes just sended by email

closed 2025-04-16 by microbotsio