microbotsio/CodeCell #13: issues with the BNO88 sensor

msinaasappel, 2025-04-08:
From version 1.2.7 there seems to be something wrong with the i2c settings for the BN088. I got the following error after loading the script.

CodeCell v1.2.8 Booting..
Initializing Sensors: E (1519) i2c.master: I2C transaction timeout detected
E (1520) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up
E (1533) i2c.master: I2C hardware timeout detected
E (1533) i2c.master: probe device timeout. Please check if xfer_timeout_ms and pull-ups are correctly set up

Error: Motion Sensor not found - Check Hardware

Additionally, the LED has turned solid red.
using 1.2.6 does not give these errors

microbotsio, 2025-04-09:
Hi

Thank you so much for reaching out

It seems we are having an issue on some devices with the v1.2.7/v1.2.8
software releases related to this matter

We are trying to fix it and push an new software release later this week
Thanks for you patiencence
[image: Logo]
Microbots Team
Miniaturized Robotics + Simplified Electronics
email: ***@***.***
www.microbots.io
[image: Twitter icon] <https://twitter.com/microbots_io>  [image: Instagram
icon] <https://www.instagram.com/microbots.io/>

microbotsio, 2025-04-09:
Hi

We've successfully replicated the issue you reported.

To help us further understand the situation, could you please confirm
whether you're using the new ESP v3.2.0 board package by Espressif?
Currently, this update isn't compatible with the latest 1.2.8 software due
to some changes in the I2C driver.

As mentioned, we're actively updating the library to ensure compatibility.
In the meantime, you can continue using the CodeCell with the previous
v3.1.3 ESP board package by Espressif.

Thank you for your patience. We'll keep you updated as soon as the upgrade
is complete.

Microbots Team

msinaasappel, 2025-04-09:
Hi,

Yes, I can confirm that I used the ESP v3.2.0

microbotsio, 2025-04-22:
We just released v1.2.9 which is now compatible with latest ESP v3.2.0 board package

Let us know if you have any other questions

closed 2025-04-22 by microbotsio