microbotsio/CodeCell #14: Button failure to when adding rotational data.

Uruglee2, 2025-04-16:
Hardware Setup:

Connect CodeCell/BNO085 sensor to the ESP32 according to standard wiring (e.g., I2C).
Connect a standard pushbutton between ESP32 GPIO Pin 6 and GND. (Pin 5 also shows the issue).
(Optional) Connect an LED to GPIO Pin 2 for status.

Problem:
When I press the button nothing happens.
Is there an issue in my code or hardware ie. the BNO085 fusion sensor?
If I write Isolation code for the LED and the button, the GPIO works however, when I introduce this to the Rotation data the button is no responsive.
The code I wrote for the BNO085 and works fine. It is just when I introduce a button press to the mess (script) I get nothing and freezes and will not let me proceed.

Question: (dumb question): previously, I was able to incorporate a button press to store the fusion sensor data at that instance, giving me a reference number. After the recent updates I have not been able to get anything working again together.

//sample code for duplicating the issue//

#include <CodeCell.h>

const int BUTTON_PIN = 6; // Pin exhibiting the issue
const int STATUS_LED_PIN = 2;
CodeCell myCodeCell;
unsigned long lastCheckTime = 0;

void setup() {
  Serial.begin(115200);
  while (!Serial && millis() < 2000);
  Serial.println("\n--- CodeCell GPIO Conflict Test ---");

  pinMode(STATUS_LED_PIN, OUTPUT);
  digitalWrite(STATUS_LED_PIN, LOW); // LED Off initially

  Serial.print("Configuring Button Pin "); Serial.println(BUTTON_PIN);
  pinMode(BUTTON_PIN, INPUT_PULLUP); // Configure button pin

  // Allow time for setup before sensor init
  Serial.println("Brief delay before sensor init...");
  delay(1000);
  int initialRead = digitalRead(BUTTON_PIN);
  Serial.print("Pin state BEFORE Init: "); Serial.println(initialRead == HIGH ? "HIGH (Expected)" : "LOW (Unexpected!)");

  Serial.println("Initializing CodeCell (MOTION_ROTATION)...");
  myCodeCell.Init(MOTION_ROTATION);
  Serial.println("CodeCell Initialization Complete.");

  // Allow time for stabilization after init
  Serial.println("Adding 1s stabilization delay...");
  delay(1000);
  Serial.println("Stabilization complete. Starting loop checks.");
  lastCheckTime = millis();
}

void loop() {
  unsigned long currentTime = millis();

  // Check button state periodically
  if (currentTime - lastCheckTime >= 500) { // Check every 500ms
    int currentButtonState = digitalRead(BUTTON_PIN);
    Serial.print("Timestamp: "); Serial.print(currentTime);
    Serial.print(" | Raw Pin "); Serial.print(BUTTON_PIN);
    Serial.print(" Read: "); Serial.println(currentButtonState);

    // Blink LED if LOW is read (to visually confirm)
    if (currentButtonState == LOW) {
        digitalWrite(STATUS_LED_PIN, !digitalRead(STATUS_LED_PIN)); // Blink LED
    } else {
        digitalWrite(STATUS_LED_PIN, LOW); // Keep LED off if HIGH
    }
    lastCheckTime = currentTime;
  }

  // Keep sensor running (even if not explicitly reading rotation data here)
  myCodeCell.Run(50); // Run at 50Hz
}

microbotsio, 2025-04-16:
Hi! Thanks for the detailed write-up - The main thing that stands out is that `myCodeCell.Init()` must be called before setting up your pins. In your current sketch, you're calling `pinMode(BUTTON_PIN, INPUT_PULLUP)` before initializing CodeCell, which is likely causing the issue. CodeCell internally configures its pins during `Init()`, and anything set beforehand may be overwritten.

Here’s a simplified version of your code that should work more reliably:

```cpp
#include <CodeCell.h>

CodeCell myCodeCell;

#define BUTTON_PIN 6U
#define STATUS_LED_PIN 2U

void setup() {
  Serial.begin(115200);
  myCodeCell.Init(MOTION_ROTATION); // Initialize CodeCell first

  // No need to call pinMode() yourself
}

void loop() {
  if (myCodeCell.Run(50)) {  // Run at 50Hz
    if (myCodeCell.pinRead(BUTTON_PIN)) {
      myCodeCell.pinWrite(STATUS_LED_PIN, HIGH);
    } else {
      myCodeCell.pinWrite(STATUS_LED_PIN, LOW);
    }
  }
}
```

Other tips that might help:
- Try using a pull-up or pull-down resistor with your button to make sure the logic level is clean and you're properly detecting rising/falling edges.
- You can skip the usual `pinMode()` setup as `myCodeCell.pinRead()` and `pinWrite()` work directly without extra initialization.
- If you want a consistent loop timing, use `myCodeCell.Run(50)` that runs the loop at 50Hz

Let us know if this works for you or if you're still seeing issues. Happy to help further!

Uruglee2, 2025-04-18:
Thank you I am taking that into consideration and I am also taking into consideration my ability to think clearly with minimal sleep. 
A simple mistake in pins is all it takes to have a couple hard days.
I appreciate the great support, Great work!!.

L

microbotsio, 2025-04-19:
No worries - we've all been there!

Glad it's getting sorted, and appreciate the kind words!

closed 2025-04-19 by microbotsio