microbotsio/CodeCell #15: Powering CodeCell using the castellated pins

alexfernandez, 2025-04-17:
Hi, I'd like to power my CodeCells using the castellated pins instead of using the JST connector. Is it possible? I guess the 5V0 pin can take 5V, but I'm using a LiPo battery in my design.

Can the CodeCell be powered using a 1S LiPo battery using the 4V4 and GND pins? I'd try it out but don't want to fry them 😅

alexfernandez, 2025-04-17:
For anyone who comes here looking for an answer: yes, you can input ~4.2 V via the VO (4.4V) and GND pins.

closed 2025-04-17 by alexfernandez

microbotsio, 2025-04-17:
Hi

Thanks for reaching out!

Yes, it will work, but we don’t recommend soldering the battery directly to
the Vo pin. That’s because Vo is regulated through the battery management
chip, not directly connected to the battery. So if you connect a battery
there, you'd be bypassing the protection and charging circuitry. When
connecting the usb this can result in incorrect charging voltage and
potentially harm the battery.

Vin, on the other hand, is directly connected to the USB charging voltage.

So while connecting a LiPo to Vo or Vin can power the CodeCell, it’s
important to note that doing so while the USB is also connected, could
damage the battery.

Our recommendation is to use the onboard connector, which ensures safe
charging and operation through the battery management chip.

Let us know if you have any other questions!

Team Microbots

alexfernandez, 2025-04-23:
Thanks a lot for the quick response! I will try using the onboard connector 😊