microbotsio/CodeCell #16: Non-void functions end without a return.

sbaeder, 2025-04-25:
Using the "latest" library in the aurduino IDE, and got "warnings" as an error on 4 routines that have an if statement that if true will return a value, but on the "else", they just print an error.

there are 4 of them... I added the "return of a zero value" on the failure to be  enabled.  That fixed the compile errors that prevented me from completing any of the examples.

Thanks

`bool CodeCell::Motion_TapRead() {
  if ((_msense & MOTION_TAP_DETECTOR) == MOTION_TAP_DETECTOR) {
    return _tap_data;
  } else {
    Serial.println(">> Error: Motion Tap Sensor not Activated");
	return 0;
  }
}

uint16_t CodeCell::Motion_StepCounterRead() {
  if ((_msense & MOTION_STEP_COUNTER) == MOTION_STEP_COUNTER) {
    return _step_data;
  } else {
    Serial.println(">> Error: Motion Step Sensor not Activated");
	return 0;
  }
}

uint16_t CodeCell::Motion_StateRead() {
  if ((_msense & MOTION_STATE) == MOTION_STATE) {
    return _mstate_data;
  } else {
    Serial.println(">> Error: Motion State Sensor not Activated");
	return 0;
  }
}

uint16_t CodeCell::Motion_ActivityRead() {
  if ((_msense & MOTION_ACTIVITY) == MOTION_ACTIVITY) {
    return _activity_data;
  } else {
    Serial.println(">> Error: Motion Activity Sensor not Activated");
	return 0;
  }
}
`

microbotsio, 2025-04-26:
Thanks for pointing this out - we really appreciate it! This has been fixed in v3.0.1.

The return statements were only missing in those cases where the sensor wasn’t properly initialized, so it shouldn't effect unless there is an error. Make sure to always initialize the sensors you're using before calling those functions. For example: myCodeCell.Init(MOTION_STEP_COUNTER);

Let us know if you still run into any issues!

closed 2025-04-26 by microbotsio

sbaeder, 2025-04-26:
thanks - Yes, it wouldn't have been an error, but compiler was flagging it as a warning that it deemed an error.  Thanks for quick response.