microbotsio/CodeCell #17: CodeCell dimensions

alexfernandez, 2025-04-27:
Hi, I'm trying to integrate CodeCell into a drone design, but I have not found an engineering diagram with all dimensions. I can measure them myself but the results are not very exact, and I cannot get good measures from the 3d model either.

* Do you have an engineering diagram for the CodeCell with all measurements?
* Failing that, what is the distance between the border and the pin centers? I have measured it to be 1.27 mm approx.
* What is the distance between the pins on opposing sides? I have measured 15.90 mm approx.

[image]

Thanks in advance!

microbotsio, 2025-04-28:
Hi 
Thanks for reaching out - the 3d model of the CodeCell is available here - https://github.com/microbotsio/CodeCell/tree/main/hardware

You can also find the measurements in the image below

[image]

Let us know if you have further questions! Happy Building :)

alexfernandez, 2025-04-29:
Thanks for your prompt answer! While those are useful, it would be great to have a proper engineering drawing: https://en.wikipedia.org/wiki/Engineering_drawing less error-prone than the 3d model and more complete than the drawing.

microbotsio, 2025-04-30:
Thanks for your message!

The 3D model is accurate, but for your convenience, please find a proper engineering drawing here:
https://github.com/microbotsio/CodeCell/blob/main/hardware/codecell%20drawing.png

closed 2025-04-30 by microbotsio

alexfernandez, 2025-04-30:
That is awesome! Thanks a lot.

For anyone coming here to integrate CodeCells into their own PCB designs, the distance between consecutive holes is 2.54 mm, and the distance between hole and border is 1.27 mm -> the distance between holes on opposing sides is 15.96 mm (18.5-1.27x2).