microbotsio/CodeCell #18: Wakeup Compilation Error

mhl787156, 2025-05-01:
Hi! 

I've been trying to setup a CodeCell today and I've gotten a weird compilation error, was wondering if you've seen this before:

```
CodeCell\src\CodeCell.cpp: In member function 'void CodeCell::USBSleep()':
CodeCell\src\CodeCell.cpp:257:45: error: 'ESP_GPIO_WAKEUP_GPIO_LOW' was not declared in this scope; did you mean 'ESP_EXT1_WAKEUP_ALL_LOW'?
  257 |   esp_deep_sleep_enable_gpio_wakeup(1 << 0, ESP_GPIO_WAKEUP_GPIO_LOW); /*Set Gpio0 as wakeup pin*/
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                             ESP_EXT1_WAKEUP_ALL_LOW
c:\Users\CodeCell\src\CodeCell.cpp:257:3: error: 'esp_deep_sleep_enable_gpio_wakeup' was not declared in this scope; did you mean 'esp_sleep_enable_gpio_wakeup'?
  257 |   esp_deep_sleep_enable_gpio_wakeup(1 << 0, ESP_GPIO_WAKEUP_GPIO_LOW); /*Set Gpio0 as wakeup pin*/
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |   esp_sleep_enable_gpio_wakeup
exit status 1

Compilation error: exit status 1
```

I have esp32 by Espressif Systems v3.2.0, CodeCell Library v1.3.1

Building on Windows 11.

microbotsio, 2025-05-02:
Hi

Thanks for reaching out

Can you share your code? can you confirm how you are using the
myCodeCell.USBSleep() function?

Thanks
Team Microbots

mhl787156, 2025-05-02:
Hi

I was using some custom code, but I just tried it with the GettingStarted Example and it raised the same issue.

To me it looks like a compilation error within the CodeCell Library itself, perhaps a version mismatch or an update in the esp32c3 library causing the enums to be switched around. 

Investigating a bit more, something is odd with my laptop setup. 
I also tried it with esp32 library v3.1.0 and v3.0.0 and it failed, Codecell library 1.2.7 and 1.3.1 also failed.

On a different machine
I have gotten it to work with CodeCell library v1.2.7/ esp32 3.2.0. A colleague also got the same. 

... very odd!

mhl787156, 2025-05-02:
Okay, got to the bottom of it, apologies it was me - I had selected `ESP32 Dev Module` instead of `ESP32C3 Dev Module`, and it was therefore compiling against the wrong firmware. Apologies all solved and hope it helps someone else out...

closed 2025-05-02 by mhl787156

microbotsio, 2025-05-02:
No worries, thanks for the update and glad you resolved the issue :)