microbotsio/CodeCell #19: Issue with GettingStarted Sketch

MomoRazor, 2025-05-20:
Hi guys.

Firstly wanted to congratulate you on how great this project looks. I wish you all the best in continuing to improve on it, and making electronics more accessible to hobbists.

I've been playing around with a CodeCell lately, and can't seem to get the GettingStarted sketch to work. The following is all the information I can think of getting, let me know if there should be more:

Error - ...\libraries\CodeCell\src\CodeCell.cpp: In member function 'void CodeCell::USBSleep()':
c:\Users\Maurovic Cachia\OneDrive\Documents\Arduino\libraries\CodeCell\src\CodeCell.cpp:257:45: error: 'ESP_GPIO_WAKEUP_GPIO_LOW' was not declared in this scope; did you mean 'ESP_EXT1_WAKEUP_ANY_LOW'?

IDE - Arduino IDE
CodeCell library version - 1.3.2
Board selected and detected by IDE - Namino Arancio

Thanks in advance for your help :)

microbotsio, 2025-05-21:
Hi

Thanks for reaching out - we're glad you like our CodeCell project!

Can you confirm you are using the latest v3.2.0 ESP32 board manager? Also
make sure you are selecting the correct board -  Tools > Board > ESP32C3
Dev Module.

More debugging information can be found here -
https://microbots.io/blogs/codecell/debugging-codecell

Let us know if this resolves the issue :)

Team Microbots

MomoRazor, 2025-05-21:
Thanks for replying. I've changed the information you mentioned and have made progress. If I can suggest something, seeing as some of the information in https://microbots.io/blogs/codecell/debugging-codecell is actually setup (namely the stuff with regards to the first Arduino IDE runs), maybe they ought to be moved to a Getting Started Tutorial, along with all the other tutorials :).

Regardless, I have made progress but I'm still running into an issue, although it does seem a bit weird and I'm not entirely sure its a CodeCell issue. Essentially, when I attempt to flash any script, for example the GettingStarted one, i receive the following issue:

```
A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
```

I've followed the troubleshooting suggestions here and have changed ports, cabels and even microcontrollers (I made sure that an other project of mine that doesn't use CodeCell was still uploading properly) and I could not find the culprit so far. I have also tried rebooting my system, and reinstalling USB drivers, but to no avail.

Not sure if you might have any insight as to what might be the issue here. I will be continuing to try and find the issue as well, and will post here if I find a solution for it.

Thank you!

MomoRazor, 2025-05-22:
Just wanted to return back to this to mention that I have managed to fix this issue, sort of.

I ended up switching the system I was working on to one that runs on Linux (Ubunt 24.04.2 LTS specifically) and everything worked properly immediately, indicating that it is probably something to do with the way my Windows system is setup somehow.

I will leave it to you to close this issue if you feel like this shouldn't be investigated any further.

MomoRazor, 2025-05-22:
Hi guys, sorry to have to re-open this issue, but I've run into some new issues, and I'm not sure if they are related to the previous issue, or some of the steps I took in attempt to fix.

So as mentioned in the previous message, I have managed to get the chip to flash new code, but I'm not sure which examples should be working and which shouldn't.

From the CodeCell examples, while Light ones seem to be working, it is clear that all Motion examples are not, but that is to be expected since my CodeCell does not have the motion sensor onboard,

Then I moved on to the CodeCell MicroLink examples, but apart from the LightSense example, which from what I gather was the example already present on the board, none seem to worki. When I successfully upload any other example, the blue LED turns off or does not blink, and the application is unable to find the CodeCell. I have not tried all the other examples, but so far all of them have had the same result.

I did run a couple of power cycles, and used your instructions to attempt to reset the chip (grounding the SCL pin), but this does not seem to change behaviour.

Thanks in advance for any advice regarding this!

microbotsio, 2025-05-24:
Hi,

Thanks again for the updates and for sharing all the detailed info!

Understood, and appreciate the suggestion about the tutorial structure, you are right, we will do our best to make it better - If your intrested we do have a YouTube video covering the initial Arduino setup here: https://www.youtube.com/watch?v=DQQVjrn1RQo&t=51s

It’s almost a year old now, but we’re working on updating it soon. But the core setup is still valid if you're looking to revisit those early steps.

The flashing issue on Windows definitely sounds strange. Since it works fine on our end (we also use Windows), it might indeed be driver-related. Reinstalling the Arduino IDE or checking for any COM port conflicts might help if you ever go back to the Windows machine. But good to hear everything’s working on Ubuntu now!

Regarding the MicroLink examples: if you’re using a CodeCell Light (without the motion sensor), then the motion examples will naturally fail - running them should throw an error over serial that the sensor is missing. That’s expected.

To get MicroLink examples running on CodeCell Light, just make sure:

1. Your setup starts with:

   ```cpp
   myCodeCell.Init(LIGHT);
   ```
2. And, remove any motion sensor calls used to update app metrics. For example, replace:

   ```cpp
   // Read rotation data from the motion sensor
   myCodeCell.Motion_RotationRead(Roll, Pitch, Yaw);

   // Show sensor readings (battery level, light sensor, yaw) in the app
   myMicroLink.ShowSensors(myCodeCell.BatteryLevelRead(), myCodeCell.Light_ProximityRead(), Yaw);
   ```

   with just:

   ```cpp
   // Show sensor readings (battery level, light sensor) in the app 
   myMicroLink.ShowSensors(myCodeCell.BatteryLevelRead(), myCodeCell.Light_ProximityRead(), 0);
   ```

Some of the examples that do not use the motion sensor, will work right away. But that should allow the other MicroLink examples (except ActivityTracker) to work properly.

Let us know how that goes
Thanks,
Team Microbots

MomoRazor, 2025-05-24:
> Regarding the Windows issue, we just noticed there was a similar issue reported a couple of weeks ago on our GitHub: [#18](https://github.com/microbotsio/CodeCell/issues/18)
> 
> Could you please confirm that in the Arduino IDE, under Tools > Board, you’ve selected ESP32C3 Dev Module and not ESP32 Dev Module?

I've retried on windows just in case I had made this mistake, but it still persisted. I do think it might be an issue with COM port conflicts, and I'll probably be investigating it further, but in any case, it does not seem to be related to CodeCell really. I will keep you posted if I find out otherwise.

> Some of the examples that do not use the motion sensor, will work right away. But that should allow the other MicroLink examples (except ActivityTracker) to work properly.

This is where I'm having issues. So far I've tested the following MicroLink examples and got the following results:

- DriverCell_Slider: Uploads ok, however bluetooth is not able to find the CodeCell, LED is on solid blue.
- FlatFlap_Slider: Uploads ok, however bluetooth is not able to find the CodeCell, LED is on solid blue.
- GettingStarted: Uploads ok, however bluetooth is not able to find the CodeCell, LED is on solid blue.
- LightSense: Uploads ok, and bluetooth connects, allowing me to add the CodeCell to the list. LED is flashing blue.
- MotorCell_Slider: After installing the MotorCell library, uploads ok, and bluetooth connects, allowing me to add the CodeCell to the list. LED is controlled by logic so not worth mentioning here.
- Motors_Joystick: After installing the DriveCell library, uploads ok, however bluetooth is not able to find the CodeCell, LED is turned off completely.

Just to be clear also, here when I say bluetooth is not able to find the CodeCell, I'm trying to add the CodeCell from scratch, having deleted when it was found using the LightSense example.

After reviewing the code, I do see two key difference in the ones that worked and the ones that did not. Its the myCodeCell.Run frequency at the start of the loop function (its 10 when it works and 100 when it doesn't) and also the myCodeCell.Init argument. While the .Run function seems to work either way, the Init seems to be the reason, and in hindsight, I might be understanding that it should be LIGHT only when i'm using a CodeCell Light, correct? I have managed to make Motors_Joystick work after changing the arguments of myCodeCel.Init to just LIGHT.

If this is the issue, I would think its important to mention in the documentation for those who have a CodeCell Light, and I would also check if this issue happens as well for people using the full CodeCell when using LightSense and MotorCell_Slider.

Thank you :)

MomoRazor, 2025-05-26:
Thanks for the help, all is working now, after doing the code changes required as you said. Not sure if the CodeCell Light is a limited run product or if its going to continue being sold, but if it is, I do suggest some specific help somewhere to make sure these code changes are more obvious to people trying out the CodeCell Light first. 

If you deem it worth it, I can create a PR with the examples that can work on the CodeCell Light, fixed to do so without user edits. I can also this on the MicroLink repo.

Thanks again for the help and for this project :)!

microbotsio, 2025-05-28:
You're very welcome - glad to hear it's working now! And thanks a lot for the thoughtful suggestion about making things clearer for CodeCell Light users. We really appreciate the feedback, and it's something we’ll keep in mind as we continue improving the docs and examples.

Thanks again for diving in and for the kind words!