microbotsio/CodeCell #20: Sleep / Wake not working

filip-dobrocky, 2025-05-31:
There seems to be a problem in the implementation of the sleep / wake function, the example `Light_SleepExample.ino` itself is not working:
```
#include <CodeCell.h>

CodeCell myCodeCell;

void setup() {
  Serial.begin(115200); 

  delay(60);//Waking up from Sleep - add a small delay for Serial
  if (myCodeCell.WakeUpCheck()) {
    // Initialize light sensor
    while (myCodeCell.Light_Init() == 1) {
      delay(1);
      myCodeCell.LightReset(); //If sensor not responding, reset it
    }
    delay(40);
    myCodeCell.Light_Read();//Read value from light sensor
    if (myCodeCell.Light_ProximityRead() < 10) {
      Serial.println("Going to sleep");
      myCodeCell.Sleep(1); //If Proxity still not detected go back to sleep & check again after 1 sec
    }
  }

  myCodeCell.Init(LIGHT); //Time to wake up - Initializes all CodeCell peripherals
  Serial.println("Initialized");
}

void loop() {
  if (myCodeCell.Run(10)) {  //Run every 10Hz
    if (myCodeCell.Light_ProximityRead() < 10) {
      Serial.println("Going to sleep");
      myCodeCell.Sleep(1); /*If Proxity not detected go to sleep & check again after 1 sec*/
    }
  }
}
```
(I just added some serial writes for debugging)

Using  Arduino IDE 2.3.6 on Linux, arduino-esp32 3.3.0-alpha1, CodeCell 1.3.2, setting Core Debug Level to "Debug", I get the following output:

```
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-C3
  Package           : 0
  Revision          : 0.04
  Cores             : 1
  CPU Frequency     : 160 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   318408 B ( 310.9 KB)
  Free Bytes        :   282768 B ( 276.1 KB)
  Allocated Bytes   :    31672 B (  30.9 KB)
  Minimum Free Bytes:   277792 B ( 271.3 KB)
  Largest Free Block:   139252 B ( 136.0 KB)
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         :  4194304 B (4 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
------------------------------------------
                nvs : addr: 0x00009000, size:    20.0 KB, type: DATA, subtype: NVS
            otadata : addr: 0x0000E000, size:     8.0 KB, type: DATA, subtype: OTA
               app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : May 31 2025 11:51:35
  Compile Host OS   : linux
  ESP-IDF Version   : v5.5-dev-3129-gd930a386da-dirty
  Arduino Version   : 3.3.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : CODECELLC3
  Arduino Variant   : codecell
  Arduino FQBN      : esp32:esp32:codecell:UploadSpeed=921600,CDCOnBoot=cdc,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=debug,EraseFlash=none,JTAGAdapter=builtin,ZigbeeMode=default
============ Before Setup End ============
[   667][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 18 already has type USB_DM (38) with bus 0x3fc917a0
[   668][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 19 already has type USB_DP (39) with bus 0x3fc917a0
 
 
>> CodeCell v1.3.2 Booting.. 
>> Initializing Sensors: [  2128][I][esp32-hal-i2c-ng.c:105] i2cInit(): Initializing I2C Master: num=0 sda=8 scl=9 freq=400000
Light Activated
[  2132][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
[  2133][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
[  2213][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
[  2293][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
[  2373][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
>> Reading Sensors: Proximity: 14361 cts | White: 823 lx | Ambient: 322 lx |
[  2455][D][esp32-hal-adc.c:272] __analogRead(): Calling __analogInit! pin = 4
Initialized
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   318408 B ( 310.9 KB)
  Free Bytes        :   278508 B ( 272.0 KB)
  Allocated Bytes   :    35244 B (  34.4 KB)
  Minimum Free Bytes:   277792 B ( 271.3 KB)
  Largest Free Block:   139252 B ( 136.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
     0 : GPIO
     1 : GPIO
     2 : GPIO
     3 : GPIO
     4 : ADC_ONESHOT[0][4]
     5 : GPIO
     6 : GPIO
     7 : GPIO
     8 : I2C_MASTER_SDA[0]
     9 : I2C_MASTER_SCL[0]
    10 : RMT_TX
    18 : USB_DM
    19 : USB_DP
    20 : UART_RX[0]
    21 : UART_TX[0]
============ After Setup End =============
>> Sampling Rate: 10Hz
[  2485][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
>> Power Status: Running from USB Power
[  2555][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
[  2655][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
[  2755][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
[  2855][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
```
...
```
Going to sleep
[  4956][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 10 is not set as GPIO. Execute digitalMode(10, OUTPUT) first.
[  4957][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
```

Then, when trying to wake up:
```
============ Before Setup End ============
[   697][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 18 already has type USB_DM (38) with bus 0x3fc917a0
[   698][I][esp32-hal-periman.c:141] perimanSetPinBus(): Pin 19 already has type USB_DP (39) with bus 0x3fc917a0
[   758][I][esp32-hal-i2c-ng.c:105] i2cInit(): Initializing I2C Master: num=0 sda=8 scl=9 freq=400000
>> Error: Light Sensor not Activated
Going to sleep
[   801][E][esp32-hal-gpio.c:176] __digitalWrite(): IO 10 is not set as GPIO. Execute digitalMode(10, OUTPUT) first.
[   801][W][esp32-hal-rgb-led.c:21] neopixelWrite(): neopixelWrite() is deprecated. Use rgbLedWrite().
```

I believe the error ` __digitalWrite(): IO 10 is not set as GPIO. Execute digitalMode(10, OUTPUT) first` is caused by this part of the `CodeCell::Sleep()` method:
```
digitalWrite(10, LOW);
pinMode(10, INPUT);
```
as the `digitalWrite` is being called without setting `pinMode` first. I tried swapping these lines, the error disappeared, but still the problem on WakeUp remains (`Error: Light Sensor not Activated`). 

Therefore, it fails to wake up, and continues going to sleep.

filip-dobrocky, 2025-05-31:
When replacing the call to initialize the light sensor by `myCodeCell.Init(LIGHT);`, it works as expected:
```
void setup() {
  Serial.begin(115200); 

  delay(60);//Waking up from Sleep - add a small delay for Serial
  if (myCodeCell.WakeUpCheck()) {
    // Initialize light sensor
    myCodeCell.Init(LIGHT);
    delay(40);
    myCodeCell.Light_Read();//Read value from light sensor
    if (myCodeCell.Light_ProximityRead() < 10) {
      Serial.println("Going to sleep");
      myCodeCell.Sleep(1); //If Proxity still not detected go back to sleep & check again after 1 sec
    }
  }

  myCodeCell.Init(LIGHT); //Time to wake up - Initializes all CodeCell peripherals
  Serial.println("Initialized");
}
```

The problem is, that the method `CodeCell::Light_Init()` does not set the class variable `_mense`, which `Light_ProximityRead()` checks for.

Honestly, the library is pretty confusing. To me it seems, that `Light_Init()` should be a private method and the initialization should only be done externally by calling the `Init()` method.

microbotsio, 2025-06-03:
Hi

Thanks you so much for reaching out and pointing this out

We are trying to recreate the problem, and will get back to you with a fix

microbotsio, 2025-06-05:
Hi,

Thanks for your patience and for pointing this out :)

We were able to replicate the issue and have resolved it in version 1.3.3

Thanks
Team Microbots